SethKitchen/RocketDesignTeamAgilePlans #10: Research High-altitude Recovery-Parachute clustering-Grid Fins-Gas Ejection

SethKitchen, 2017-08-14:
Who: Special squad mix of Prop, Electronics, and D&M
What: Look into how to recover our rockets when they go in low oxygen areas.
When: 9/3-9/17
Where: Internet and Meetups
Why: We can't have our rocket just falling to the ground and our normal blackpowder ejection won't work because low oxygen.
How: Google keywords in this issue title and read articles about them. Become an expert. Come up with a plan we can implement on a tight budget and you'd be confident in building as a team.

daniel-klinger, 2017-08-16:
Some links:
Altimeter Comparison: http://www.rocketryforum.com/showthread.php?121323-Altimeters-Comparison-guide
Legality of Control Surfaces: http://www.rocketryforum.com/showthread.php?32665-Is-it-legal-for-me-to-use-an-active-control-system-on-my-high-powered-rocket

SethKitchen, 2017-09-12:
The plan for piston ejection right now is taking a canister of CO2 like in a paintball gun and putting it underneath the nose cone. Then we just need an altitude check from some person working on altimeters. When they say go, somehow we need to open the canister. I was originally thinking a solenoid valve, but that may be too heavy. Hopefully you can perfect this plan. Feel free to change anything/everything. Some questions would be: what's the lightest weight solenoid valve you can find? What does it take to power it and open it? Is there any other option you can think of to puncture a CO2 canister?

closed 2017-10-18 by SethKitchen