SethKitchen/RocketDesignTeamAgilePlans #3: Training in General Safety Training

SethKitchen, 2017-08-14:
Who: Everyone on RDT
What: Finish General Safety Training through Canvas or through the SDELC.
When: 9/3-9/17
Where: SDELC or Internet
Why: To access the SDELC and Work Areas
How: Sign up on orgsync or canvas, show up and listen to Richard, or do the requirements online.

SethKitchen, 2017-09-01:
Leave a little congrats to yourself when you finish :)

"Go me, I have General Safety!"

davinschmied, 2017-09-01:
Go me I have gen safety, basic machine shop, and composites!

Calebg720, 2017-09-01:
Go me, I have General Safety!

AndyManson, 2017-09-01:
Go me I have general safety

dbfv8, 2017-09-01:
Go me, i have general safety

willziehm, 2017-09-02:
go me, i have general safety!

kczn6, 2017-09-02:
Go me, I have general safety training!

jng6w3, 2017-09-02:
Go me, I have general safety training

uabwd2, 2017-09-02:
Go me, I have General Safety!

jawwm9, 2017-09-03:
Go me, I have General Safety!

Josiah-Wade, 2017-09-04:
Hip hip hooray 
Hip hip hooray
I got general safety out of the way

SethKitchen, 2017-09-04:
Nice poetry to my ears

krshg6, 2017-09-06:
Go me, I have General Safety!

trevoraldenberg, 2017-09-07:
Go me, I have General Safety!

miamagby, 2017-09-07:
Go me, I have General Safety!

bcmt42, 2017-09-07:
Go me, I have General Safety!

zjgt37, 2017-09-07:
Go me, I have General Safety!

bjg5k2, 2017-09-07:
Woo woo I did a general safety
I said that right didn't I?

djcvf, 2017-09-11:
Go me, I did a general safety

drw994, 2017-09-11:
Go me, i already have general safety and basic machine

kgpat, 2017-09-12:
Go me, I have general safety, basic machine, and composites.

SethKitchen, 2017-09-14:
By now everyone should have general safety. If you do not, you are officially behind. Complete it on canvas ASAP.

closed 2017-09-14 by SethKitchen