SethKitchen/RocketDesignTeamAgilePlans #17: Send Me Parts Needed For Midwest

SethKitchen, 2017-08-16:
Who: Daniel Klinger to Seth
What: Send me any parts you need for midwest power (ie radio and receiver)
When: Before First Meeting
Where: Internets
Why: Trying to get a budget for to give to Jeff

daniel-klinger, 2017-08-16:
RF Transmitter Module
http://store.rfdesign.com.au/rfd-900p-modem/

SethKitchen, 2017-09-01:
Transmitter=99.50,
Mixer= 750 New but rocketry forum $200, 
Resin | 105-1G (126 oz.) | 01-00354 | $98.26
Fast Hardener | 205-QT (27 oz.) | 01-00357 | $52.70
Slow Hardener | 206-QT (27 oz.) | 01-00360 | $44.29

SethKitchen, 2017-09-01:
$350 strain gauges
$600 propellant

SethKitchen, 2017-09-01:
~1750 total

closed 2017-09-01 by SethKitchen

CRobless, 2017-09-01:
You forgot to add the cost for motor parts. ~250 for 54mm testing, ~300 for nikomedes

SethKitchen, 2017-09-01:
Good call thanks