SethKitchen/RocketDesignTeamAgilePlans #19: Create Project Reveal Video

SethKitchen, 2017-09-01:
Who: Matt and members of PR group
What: Take designs and pictures from first meetings to create a video which will attract sponsors and promote image of the team.
When: 9/3-9/17
Where: Should be taken at the SDELC and through online media. Should be sent to Aerospace companies not competing with Interstellar and post on social media pages.
Why: It is important people know who we are so when we apply to launch sites they are okay with whatever we are launching. We want to spread buzz so people want to join our team. We want it to be a BIG deal when we break the collegiate record

SethKitchen, 2017-09-01:
Matt said: Plan project reveal video
-need to wait for M&D, PROP, and ELE to finalize designs
-need leadership to finalize launch date plans
-need 3rd parties to confirm and accept launch plans

SethKitchen, 2017-09-01:
I would say the designs should be finalized to a point we could make a video. What else do you need.

Tentative launch dates are Midwest Power, Late February, Late April, and  Spaceport America Cup.

Do we need them to confirm to make the video?

msf6fd, 2017-09-01:
I think we need solid confirmation from all 3rd parties that we need to launch (qcrs up in Princeton, the argonia rocket group if we intend to use them as a 30K test launch, and then Spaceport that we are good to try 70K and 150K. They are all going to have their own requirements for us to launch that we still need to incorporate into the design. I'm meeting with Dan and Daniel Monday night to try and shore up the electronics side of things. Seth you and I should meet Tuesday night or some other time and right up a project overview that we can send third parties and get feedback.

msf6fd, 2017-09-01:
Once we have all that I'd say we are set on the path of really making this happen, we do the reveal. We've unveiled rockets two years in a row that looked nothing like their final product which I think is kind of embarrassing

SethKitchen, 2017-09-01:
I've emailed QCRS and Argonia. Still in contact with Mark Twain Forest to get our own pad. Tuesday Night would work for me after 6. I understand not wanting to display something that won't happen, but we need to get the word out as soon as possible. This is the time when people are planning their donations.

SethKitchen, 2017-09-14:
Update on this?

msf6fd, 2017-09-14:
Sanders and I are meeting this weekend to block out the video

SethKitchen, 2017-09-28:
Any update on this?